DIGImend/digimend-kernel-drivers #637: EasyPen i405X stopped working after upgrade from ubuntu 20.04 to 22.04.1

jdimit, 2022-09-25:
Hello there. I am having a problem with my easypen i405x which stopped working after an upgrade to ubuntu 22.04.1. Although the device is seen in the usb list, xinput and evtest cannot see it. The digimend-dkms and dkms drivers are already the newest versions. Can anyone help to get the pen working again?

$ lsusb
Bus 003 Device 071: ID 0458:5010 KYE Systems Corp. (Mouse Systems) EasyPen i405X

$ xinput list
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ xwayland-pointer:16                     	id=6	[slave  pointer  (2)]
⎜   ↳ xwayland-relative-pointer:16            	id=7	[slave  pointer  (2)]
⎜   ↳ xwayland-pointer-gestures:16            	id=8	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ xwayland-keyboard:16                    	id=9	[slave  keyboard (3)]

$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Lid Switch
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Video Bus
/dev/input/event5:	SynPS/2 Synaptics TouchPad
/dev/input/event6:	BTC USB Multimedia Keyboard
/dev/input/event7:	BTC USB Multimedia Keyboard System Control
/dev/input/event8:	HDA ATI HDMI HDMI/DP,pcm=3
/dev/input/event9:	BTC USB Multimedia Keyboard Consumer Control
/dev/input/event10:	UVC Camera (046d:09a4)
/dev/input/event11:	HD-Audio Generic Mic
/dev/input/event12:	HD-Audio Generic Headphone
/dev/input/event13:	Bluetooth 3.0 Mouse
/dev/input/event14:	Wireless hotkeys
/dev/input/event15:	HP WMI hotkeys
Select the device event number [0-15]: 

$ sudo apt install ./digimend/digimend-dkms_11_all.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'digimend-dkms' instead of './digimend/digimend-dkms_11_all.deb'
digimend-dkms is already the newest version (11).

$ sudo apt install dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (2.8.7-2ubuntu2).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.

xTibor, 2023-03-11:
This change broke i405X support: https://github.com/DIGImend/digimend-kernel-drivers/commit/ec5c16d631940546ddaf03eecec082865bd9c676#diff-ea5625a328f4b141c37b709d8db642b2d684dcfc2997a1553bd370daaa19b369L40-R40

Reverting it makes my i405X work again.

facundopicco, 2023-09-29:
Hi there. 
I think I have the same problem. I tried to follow the steps, but still cannot get it to work.

I can see the device in lsusb
> lsusb
>> [...] Bus 005 Device 028: ID 0458:5010 KYE Systems Corp. (Mouse Systems) EasyPen i405X [...]

I have a rule working, because I get the right permissions
> ls -l /dev/bus/usb/005/028
>> crw-rw-rw- 1 root root 189, 539 Sep 29 13:42 /dev/bus/usb/005/028

I have installed the driver with the following steps:
>git clone git@github.com:DIGImend/digimend-kernel-drivers.git
>cd digimend-kernel-drivers
>git checkout 404cb12a0eff99ac3f7360c4864308d551f78afc hid-kye.c
>sed -i 's/MODULE_VERSION("10");/MODULE_VERSION("11");/' hid-kye.c
>sudo apt-get install -y "linux-headers-$(uname -r)"
>sudo apt-get install -y dkms
>sudo make dkms_install

When I run usb-devices I don't see it in the list
> usb-devices
>> I see a whole bunch of devices, but no my Genius Easypen

I am on ubuntu 22.04.2 LTS

I also see that when I move the pen on top of the tablet, the green LED turns on. But I cannot get it to do anything. 

Any idea on what else could be wrong?
Thanks!

jdimit, 2023-09-29:
Ultimately, it was a corrupted upgrade. I had to revert back to the 20.04 an everything worked fine.
Regards,
Jordan

git clone ***@***.******@***.***>:DIGImend/digimend-kernel-drivers.git
git checkout 404cb12<https://github.com/DIGImend/digimend-kernel-drivers/commit/404cb12a0eff99ac3f7360c4864308d551f78afc> hid-kye.c