DIGImend/digimend-kernel-drivers #638: Input/eventX device not created for XP-Pen Star G640

lxyd, 2022-10-09:
Hello! Thank you for great project. There is a problem with G640 tablet.

**OS:** Fedora release 36

**Kernel version:** 5.19.13-200.fc36.x86_64

**Digimend-kernel-drivers:** v11 installed with `sudo make dkms_install`

**The situation:** tablet XP-Pen Star G640 worked fine on Fedora's 5.17 kernel *without* digimend driver installed but it stopped working after update to 5.19. So I tried installing digimend-kernel-drivers but no help, see debug log later.

A couple of facts to consider:
- tablet works fine under Windows and also under Linux with proprietary drivers from xp-pen;
- tablet works kinda-fine under Linux with older kernel 5.17 without digimend;
- after installing digimend on kernel 5.17 tablet stops working;

Kernel 5.19 *with* and *without* digimend drivers show almost identical simptomes:
- lines like `uclogic 0003:28BD:0094.0007: interface is invalid, ignoring` in dmesg log;
- hidraw device is created and is readable with `sudo usbhid-dump -d 28bd -es`;
- but no input/eventX device is created, so tablet is not visible to Gnome and applications.

Dmesg log when connecting the device with debug enabled:
```
[  104.857754] usb 1-9: new full-speed USB device number 2 using xhci_hcd
[  105.145262] usb 1-9: New USB device found, idVendor=28bd, idProduct=0094, bcdDevice= 0.00
[  105.145270] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  105.145272] usb 1-9: Product: STAR G640
[  105.145274] usb 1-9: Manufacturer: XP-PEN
[  105.171115] hid_uclogic:uclogic_probe:223: uclogic 0003:28BD:0094.0007: parameters:
[  105.171120] hid_uclogic:uclogic_params_hid_dbg:117: uclogic 0003:28BD:0094.0007: .invalid = true
[  105.171123] hid_uclogic:uclogic_params_hid_dbg:119: uclogic 0003:28BD:0094.0007: .desc_ptr = 0000000000000000
[  105.171125] hid_uclogic:uclogic_params_hid_dbg:120: uclogic 0003:28BD:0094.0007: .desc_size = 0
[  105.171127] hid_uclogic:uclogic_params_hid_dbg:121: uclogic 0003:28BD:0094.0007: .pen = {
[  105.171129] hid_uclogic:uclogic_params_pen_hid_dbg:60: uclogic 0003:28BD:0094.0007: 	.usage_invalid = false
[  105.171131] hid_uclogic:uclogic_params_pen_hid_dbg:62: uclogic 0003:28BD:0094.0007: 	.desc_ptr = 0000000000000000
[  105.171133] hid_uclogic:uclogic_params_pen_hid_dbg:63: uclogic 0003:28BD:0094.0007: 	.desc_size = 0
[  105.171135] hid_uclogic:uclogic_params_pen_hid_dbg:64: uclogic 0003:28BD:0094.0007: 	.id = 0
[  105.171137] hid_uclogic:uclogic_params_pen_hid_dbg:65: uclogic 0003:28BD:0094.0007: 	.subreport_list = {
[  105.171139] hid_uclogic:uclogic_params_pen_hid_dbg:67: uclogic 0003:28BD:0094.0007: 		{0x00, 0},
[  105.171141] hid_uclogic:uclogic_params_pen_hid_dbg:67: uclogic 0003:28BD:0094.0007: 		{0x00, 0},
[  105.171143] hid_uclogic:uclogic_params_pen_hid_dbg:67: uclogic 0003:28BD:0094.0007: 		{0x00, 0}
[  105.171145] hid_uclogic:uclogic_params_pen_hid_dbg:72: uclogic 0003:28BD:0094.0007: 	}
[  105.171147] hid_uclogic:uclogic_params_pen_hid_dbg:73: uclogic 0003:28BD:0094.0007: 	.inrange = normal
[  105.171149] hid_uclogic:uclogic_params_pen_hid_dbg:75: uclogic 0003:28BD:0094.0007: 	.fragmented_hires = false
[  105.171151] hid_uclogic:uclogic_params_pen_hid_dbg:77: uclogic 0003:28BD:0094.0007: 	.tilt_y_flipped = false
[  105.171153] hid_uclogic:uclogic_params_hid_dbg:123: uclogic 0003:28BD:0094.0007: 	}
[  105.171154] hid_uclogic:uclogic_params_hid_dbg:124: uclogic 0003:28BD:0094.0007: .frame_list = {
[  105.171156] hid_uclogic:uclogic_params_hid_dbg:126: uclogic 0003:28BD:0094.0007: 	{
[  105.171158] hid_uclogic:uclogic_params_frame_hid_dbg:92: uclogic 0003:28BD:0094.0007: 		.desc_ptr = 0000000000000000
[  105.171160] hid_uclogic:uclogic_params_frame_hid_dbg:93: uclogic 0003:28BD:0094.0007: 		.desc_size = 0
[  105.171161] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:28BD:0094.0007: 		.id = 0
[  105.171163] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:28BD:0094.0007: 		.suffix = (null)
[  105.171165] hid_uclogic:uclogic_params_frame_hid_dbg:96: uclogic 0003:28BD:0094.0007: 		.re_lsb = 0
[  105.171167] hid_uclogic:uclogic_params_frame_hid_dbg:97: uclogic 0003:28BD:0094.0007: 		.dev_id_byte = 0
[  105.171169] hid_uclogic:uclogic_params_frame_hid_dbg:98: uclogic 0003:28BD:0094.0007: 		.touch_byte = 0
[  105.171171] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:28BD:0094.0007: 		.touch_max = 0
[  105.171173] hid_uclogic:uclogic_params_frame_hid_dbg:100: uclogic 0003:28BD:0094.0007: 		.touch_flip_at = 0
[  105.171175] hid_uclogic:uclogic_params_frame_hid_dbg:102: uclogic 0003:28BD:0094.0007: 		.bitmap_dial_byte = 0
[  105.171177] hid_uclogic:uclogic_params_hid_dbg:128: uclogic 0003:28BD:0094.0007: 	},
[  105.171178] hid_uclogic:uclogic_params_hid_dbg:126: uclogic 0003:28BD:0094.0007: 	{
[  105.171180] hid_uclogic:uclogic_params_frame_hid_dbg:92: uclogic 0003:28BD:0094.0007: 		.desc_ptr = 0000000000000000
[  105.171182] hid_uclogic:uclogic_params_frame_hid_dbg:93: uclogic 0003:28BD:0094.0007: 		.desc_size = 0
[  105.171184] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:28BD:0094.0007: 		.id = 0
[  105.171185] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:28BD:0094.0007: 		.suffix = (null)
[  105.171187] hid_uclogic:uclogic_params_frame_hid_dbg:96: uclogic 0003:28BD:0094.0007: 		.re_lsb = 0
[  105.171189] hid_uclogic:uclogic_params_frame_hid_dbg:97: uclogic 0003:28BD:0094.0007: 		.dev_id_byte = 0
[  105.171191] hid_uclogic:uclogic_params_frame_hid_dbg:98: uclogic 0003:28BD:0094.0007: 		.touch_byte = 0
[  105.171193] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:28BD:0094.0007: 		.touch_max = 0
[  105.171194] hid_uclogic:uclogic_params_frame_hid_dbg:100: uclogic 0003:28BD:0094.0007: 		.touch_flip_at = 0
[  105.171196] hid_uclogic:uclogic_params_frame_hid_dbg:102: uclogic 0003:28BD:0094.0007: 		.bitmap_dial_byte = 0
[  105.171198] hid_uclogic:uclogic_params_hid_dbg:128: uclogic 0003:28BD:0094.0007: 	},
[... 219 lines cut ...]
[  105.207994] hid_uclogic:uclogic_params_frame_hid_dbg:93: uclogic 0003:28BD:0094.0009: 		.desc_size = 0
[  105.207995] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:28BD:0094.0007: 		.id = 0
[  105.207995] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:28BD:0094.0009: 		.id = 0
[  105.207996] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:28BD:0094.0007: 		.suffix = (null)
[  105.207996] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:28BD:0094.0009: 		.suffix = (null)
[  105.207997] hid_uclogic:uclogic_params_frame_hid_dbg:96: uclogic 0003:28BD:0094.0007: 		.re_lsb = 0
[  105.207997] hid_uclogic:uclogic_params_frame_hid_dbg:96: uclogic 0003:28BD:0094.0009: 		.re_lsb = 0
[  105.207998] hid_uclogic:uclogic_params_frame_hid_dbg:97: uclogic 0003:28BD:0094.0007: 		.dev_id_byte = 0
[  105.207998] hid_uclogic:uclogic_params_frame_hid_dbg:97: uclogic 0003:28BD:0094.0009: 		.dev_id_byte = 0
[  105.207999] hid_uclogic:uclogic_params_frame_hid_dbg:98: uclogic 0003:28BD:0094.0007: 		.touch_byte = 0
[  105.207999] hid_uclogic:uclogic_params_frame_hid_dbg:98: uclogic 0003:28BD:0094.0009: 		.touch_byte = 0
[  105.208000] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:28BD:0094.0007: 		.touch_max = 0
[  105.208000] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:28BD:0094.0009: 		.touch_max = 0
[  105.208001] hid_uclogic:uclogic_params_frame_hid_dbg:100: uclogic 0003:28BD:0094.0007: 		.touch_flip_at = 0
[  105.208001] hid_uclogic:uclogic_params_frame_hid_dbg:100: uclogic 0003:28BD:0094.0009: 		.touch_flip_at = 0
[  105.208002] hid_uclogic:uclogic_params_frame_hid_dbg:102: uclogic 0003:28BD:0094.0007: 		.bitmap_dial_byte = 0
[  105.208002] hid_uclogic:uclogic_params_frame_hid_dbg:102: uclogic 0003:28BD:0094.0009: 		.bitmap_dial_byte = 0
[  105.208003] hid_uclogic:uclogic_params_hid_dbg:128: uclogic 0003:28BD:0094.0007: 	}
[  105.208003] hid_uclogic:uclogic_params_hid_dbg:128: uclogic 0003:28BD:0094.0009: 	}
[  105.208004] hid_uclogic:uclogic_params_hid_dbg:131: uclogic 0003:28BD:0094.0007: }
[  105.208004] hid_uclogic:uclogic_params_hid_dbg:131: uclogic 0003:28BD:0094.0009: }
[  105.208004] uclogic 0003:28BD:0094.0007: interface is invalid, ignoring
[  105.208005] uclogic 0003:28BD:0094.0009: interface is invalid, ignoring

```

nic3-14159, 2022-10-09:
Hello! I have a suspicion that this might be caused by #550. 
Could you either try applying the code in #553, or run `git revert ec5c16d631940546ddaf03eecec082865bd9c676` then build and install Digimend, and then report if the tablet works?

lxyd, 2022-10-09:
Tried both ways and both did successfully made the tablet to work. Thank you!

nic3-14159, 2022-10-09:
Great! Let's keep this issue open until either #553 is merged or more ideally, the issue gets fixed in the kernel. I'll see what I can do about that.

lukeovalle, 2022-12-21:
I had the same problem and using #553 fixed it