DIGImend/digimend-kernel-drivers #641: XP-Pen Deco 01 V2 stops working when I press the upper pen button

dcarrera, 2022-11-20:
I have an XP-Pen Deco 01 V2 and I have a weird and frustrating problem. The tablet generally works well. I can draw on Krita, pressure sensitivity seems to work, etc. The stylus has two buttons. The lower one acts like a mouse middle click button. The upper one makes the tablet stop working entirely.

To "fix" the tablet, I can unplug it and plug it back in, and then it goes back to normal.

I have created and attached all of the files that were requested in the "Collecting tablet diagnostics" page.

* [descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10048308/descriptors.txt)

* [hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10048307/hid_report_descriptors.txt)

* [probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10048305/probe.txt)

* [pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10048306/pen_buttons.txt)

While producing these files I noticed two more surprising things:

1. When I ran the 'uclogic-probe' command the stylus stopped working again, even if I didn't press the button. I 'fixed' that again by unplugging the USB and plugging it back in.
2. When I ran the 'usbhid-dump' command and pressed the buttons, the tablet continued working normally. So whatever is causing the tablet to stop working is not something inside the tablet.

I am running Pop! OS 22.04 with Gnome and X11.

Thanks for the help.

JoseExposito, 2022-12-08:
Hi!

While the XP-PEN Deco 01 seems to be supported by DIGImend, I don't think that your tablet (v2) is supported. It has a different product ID and I don't see it listed.

You can confirm it by connecting your tablet and running `dmesg`. If you see something like:

```
[ 5576.132879] input: <your tablet name> as /devices/...
[ 5576.187009] hid-generic ...
                            ^ This is the name of the driver
```

It means that your tablet is handled by the `hid-generic` driver and not by DIGImend (`uclogic` driver). Feel free to share the log and I'll have a look.

I added support for a bunch of XP-Pen devices on the kernel lately. I don't have enough information from your log to be 100% sure, but I think that supporting your tablet should be straightforward.

Could you test these drivers, please?

First, download the source code:  [digimend-kernel-drivers.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/10187349/digimend-kernel-drivers.zip)

The source code contains the code from DIGImend, the changes I made in the [upstream kernel](https://github.com/torvalds/linux/blob/master/drivers/hid/hid-uclogic-params.c) plus the changes required to add support for your tablet.

In order to install it, you'll need to install the kernel headers:

```
$ sudo apt-get install -y "linux-headers-$(uname -r)"
```

After extracting the code, cd into the source code and:

```
$ make
$ sudo make install
```

You can uninstall it running:

```
$ sudo make uninstall
```

More detailed instructions are available in the [DIGImend README file](https://github.com/DIGImend/digimend-kernel-drivers#digimend-kernel-drivers).

After installing the drivers and reconnecting, `dmesg` should show that the `uclogic` driver is used for your device and, hopefully, your pen's buttons should work. Let me know how it goes!

JoseExposito, 2022-12-26:
For reference, this issue was reported fixed and tested here: https://gitlab.freedesktop.org/libinput/libinput/-/issues/839

The patch is already in the kernel mailing list: https://lore.kernel.org/linux-input/20221226123456.14822-1-jose.exposito89@gmail.com/T/

linuxgirl22, 2023-04-25:
Hi, thanks to the awesome digimend drivers, I run the Deco 01 v2. No doubt the patches have fixed the original issue, but just to add something that might be useful regarding right click/upper button, for both v1/2 of the Deco 01. I'd been able to piece together, when first starting to use it, how to add right-click, from looking at threads here, but can't find the thread where I detailed that. Currently running MX Linux, X11, with Mate desktop installed. I'm running 5.10 kernel but not a recent version. 

On looking through the notes I keep, with not being very technical, lol, have noted that am running kernel/headers, digimend/dkms, libwacom and libevdev stuff, then add 2 files in 2 locations to enable the right click [e.g. colour pick] via top button. 

**10-xppen.hwdb into /etc/udev/hwdb.d** ... 

evdev:input:b0003v28BDp0905*
 KEYBOARD_KEY_d0045=332
 KEYBOARD_KEY_700e0=fn
 KEYBOARD_KEY_700e2=fn
 KEYBOARD_KEY_70015=256
 KEYBOARD_KEY_70018=257
 KEYBOARD_KEY_7001c=258
 KEYBOARD_KEY_7003c=259
 KEYBOARD_KEY_7002d=260
 KEYBOARD_KEY_70026=261
 KEYBOARD_KEY_70067=262
 KEYBOARD_KEY_70066=263

and  

**xp-pen-dec01.tablet into /usr/share/libwacom**

# XP-Pen
# Deco 01
#

[Device]
Name=XP-Pen Deco 01
ModelName=Deco01v2
DeviceMatch=usb:28bd:0905
Class=Bamboo
Width=10
Height=6
IntegratedIn=
Layout=xp-pen-star03.svg
Styli=0xffffd;

[Features]
Stylus=true
Reversible=true
Touch=false
Buttons=8

[Buttons]
Left=A;B;C;D;E;F;G;H;
EvdevCodes=0x100;0x101;0x102;0x103;0x104;0x105;0x106;0x107

In Krita, I set tablet settings to 'use mouse events for right/middle clicks', and also have a note about deleting the Alt+Space shortcut in Mate, but I think the latter note I made was when I was running Sai2 via wine.