DIGImend/digimend-kernel-drivers #646: Genius EasyPen i405x not working on Fedora 34

estebandres, 2022-12-27:
Kernel version 5.17.12
X11

Hello. The tablet works fine on Ubuntu 20.04 and in windows 10.
I've followed the steps described in the next link https://digimend.github.io/support/howto/trbl/diagnostics/ and I get correct readings same as listed in https://github.com/DIGImend/tablets/tree/master/KYE_EasyPen_i405X.
I've installed the digimend-kernel-drivers as suggested in the README usign DKMS. No errors reported.
Then restarted the computer just in case but nothing.

What else should I try to get it to work?

nic3-14159, 2022-12-27:
I have a quick theory that it may have something to do with the same issue in #550. As a quick test, could you try running `git revert ec5c16d631`, reinstall digimend, reboot, and see if it works?

estebandres, 2022-12-27:
You are absolutely right. I did what you asked and now the tablet works perfectly. Thank you!