DIGImend/digimend-kernel-drivers #650: Huion H610 PRO V2 stops working after indeterminate amount of time

ssterling, 2023-01-26:
Upon being plugged in, or upon starting my window manager, my Huion H610 PRO V2 tablet works as expected only for an indeterminate amount of time, ranging from under five seconds to nearly ten minutes.  This happens regardless of whether X11 is running.

Whenever the tablet stops working, it still appears to be connected in kernel logs, `lshw`, `ls /dev/input/event*` and `xinput list`.  The green LED on the tablet continues to work as normal, lighting up when the pen is close to the pad or if a button is being pressed.  The only way to reset the tablet is to physically disconnect and reconnect the tablet.

I’m using the mainline `hid_uclogic` kernel module as of 6.1.5, and, when using X11, [xf86-input-wacom](https://github.com/linuxwacom/xf86-input-wacom).

Following are evtest and kernel logs from one instance in which all I did was reconnect it and press the Undo button (Button 1 on the pad) twice before it stopped working:
```console
smp@sakura:~ # sudo evtest /dev/input/event26
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6d version 0x111
Input device name: "HUION Huion Tablet Pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 257 (BTN_1)
    Event code 258 (BTN_2)
    Event code 259 (BTN_3)
    Event code 260 (BTN_4)
    Event code 261 (BTN_5)
    Event code 262 (BTN_6)
    Event code 263 (BTN_7)
    Event code 264 (BTN_8)
    Event code 265 (BTN_9)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 306 (BTN_C)
    Event code 331 (BTN_STYLUS)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1674753754.558298, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1674753754.558298, type 1 (EV_KEY), code 256 (BTN_0), value 1
Event: time 1674753754.558298, -------------- SYN_REPORT ------------
Event: time 1674753754.684219, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1674753754.684219, type 1 (EV_KEY), code 256 (BTN_0), value 0
Event: time 1674753754.684219, -------------- SYN_REPORT ------------
```
```console
smp@sakura:~ # sudo tail -f -n 0 /var/log/syslog
Jan 26 12:22:31 sakura kernel: usb 3-2: new full-speed USB device number 25 using xhci_hcd
Jan 26 12:22:31 sakura kernel: usb 3-2: New USB device found, idVendor=256c, idProduct=006d, bcdDevice= 1.00
Jan 26 12:22:31 sakura kernel: usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 26 12:22:31 sakura kernel: usb 3-2: Product: Huion Tablet
Jan 26 12:22:31 sakura kernel: usb 3-2: Manufacturer: HUION
Jan 26 12:22:31 sakura kernel: input: HUION Huion Tablet Pen as /devices/pci0000:00/0000:00:01.3/0000:03:00.2/0000:20:08.0/0000:26:00.0/usb3/3-2/3-2:1.0/0003:256C:006D.0034/input/input137
Jan 26 12:22:31 sakura kernel: input: HUION Huion Tablet Pad as /devices/pci0000:00/0000:00:01.3/0000:03:00.2/0000:20:08.0/0000:26:00.0/usb3/3-2/3-2:1.0/0003:256C:006D.0034/input/input138
Jan 26 12:22:31 sakura kernel: input: HUION Huion Tablet Touch Strip as /devices/pci0000:00/0000:00:01.3/0000:03:00.2/0000:20:08.0/0000:26:00.0/usb3/3-2/3-2:1.0/0003:256C:006D.0034/input/input139
Jan 26 12:22:31 sakura kernel: input: HUION Huion Tablet Dial as /devices/pci0000:00/0000:00:01.3/0000:03:00.2/0000:20:08.0/0000:26:00.0/usb3/3-2/3-2:1.0/0003:256C:006D.0034/input/input140
Jan 26 12:22:31 sakura kernel: uclogic 0003:256C:006D.0034: input,hiddev97,hidraw6: USB HID v1.11 Keypad [HUION Huion Tablet] on usb-0000:26:00.0-2/input0
Jan 26 12:22:31 sakura kernel: uclogic 0003:256C:006D.0035: No inputs registered, leaving
Jan 26 12:22:31 sakura kernel: uclogic 0003:256C:006D.0035: hidraw7: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:26:00.0-2/input1
```

Jayanth-L, 2023-12-25:
Same issue here, were yo able to find the solution ?

ssterling, 2024-01-24:
No, unfortunately.