DIGImend/digimend-kernel-drivers #651: Rii Graphics Tablet X Value Capped at 25400, Only 1/3 of the tablet area is usable.

Real-Time-Kodi, 2023-01-27:
The pen works perfectly on the Y-axis, but when it's used on the X-axis, the entire width of the screen is traversed when the pen has only moved about 1/3 of the way across the tablet. All of the tests I've tried seem to show the x value is capped as 25400(you can see it in the pen_coords.txt as 38 63 in the third and fourth place in each HID packet)

I've tried looking at the USB interface on a windows PC and I can't seem to find any consistent communication between the driver package and the tablet. The buttons on the tablet show up as a separate USB keyboard endpoint.

Pressure seems to be working properly.

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10515204/descriptors.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10515205/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10515206/pen_pressure.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/10515207/probe.txt)

Real-Time-Kodi, 2023-03-01:
[pcap2.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/10857267/pcap2.zip)
I did some packet capture looking for an initialization packet. It appears that 25400 is the proper maximum value, and both the windows and linux are being given the same descriptors. I am seeing a couple of windows-specific descriptor packets that wireshark can't handle.

Also, on windows there are SET_REPORT and GET_REPORT packets being sent to the tablet while the tablet's configuration software is running. Preventing the software from starting doesn't seem to affect the tablet though, it works either way and those packets aren't present when the software isn't running.

I'm a little stumped at this point. I'll continue to work on it but I'd appreciate it if anyone has any insight.

guyuming76, 2024-04-24:
I get similar issue with my GaoMon tablet 1060 pro: only left 1/3 of the pad is usable.

I don't have any capture of any kind.