DIGImend/digimend-kernel-drivers #653: incompatible tool with wayland

DrDeaDroID, 2023-02-04:
when I try to enter `xsetwacom` command, I get an error about wayland incompatibility
```
$ xsetwacom --list
Wayland devices found but this tool is incompatible with Wayland. See
https://github.com/linuxwacom/xf86-input-wacom/wiki/Wayland
```

gelatinbomb, 2023-05-20:
Well xsetwacom is for X.Org so I don't think you can't do much about it. But hey as far my experience go with wayland at least on the latest versions it's quite "plug and play" with graphic tablets. My Xp Pen Artist 15.6 Pro runs by default there. If you use kde on the lastest versions it provides a panel to set up some really basics funcions of the tablet like monitor mapping. It still lacks some important tools like callibration or express keys mapping.

yaitskov, 2024-09-07:
oh