DIGImend/digimend-kernel-drivers #661: Huion Inspiroy 2 Support

jacobranson, 2023-06-16:
I specifically have the 2 M. Unsure if there would be a separate effort for the large and small variants. Happy to help debug, or even throw a few bucks in to make this happen.

raman-maker, 2023-11-22:
Are you able to use Huion Inspiroy 2 tablet ? 
is your problem solved ?

jacobranson, 2023-11-22:
@raman-maker There hasn't been a commit on main in 7 months, so no.

raman-maker, 2023-11-23:
HUION Inspiroy 2 tablet works on Manjaro Linux.  Use command below to install it :
`pamac build huiontablet`

https://forum.manjaro.org/t/huion-inspiroy-2-driver-installation-in-manjaro-linux/152049/1

jacobranson, 2023-11-23:
Yeah but that's not part of the digimend kernel drivers. That's an out of tree driver that isn't open source. @raman-maker

JoseExposito, 2024-05-03:
Hi!

Your issue should be fixed by [DIGImend v13](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v13). Please, if you have a chance, give it a try and report any problems found.

This was the related pull request fixing it: https://github.com/DIGImend/digimend-kernel-drivers/pull/679

closed 2024-05-03 by JoseExposito