DIGImend/digimend-kernel-drivers #669: Huion Kamvas 16 touch strip not working

matitaelettrica, 2023-10-13:
I have a huion Kamvas 16 pen display model GS1561, installed latest version to me available, v.11
Linux Mint 21.2 Cinnamon, kernel 5.15.0-86-generic
[Apologies for wonky English, ESL here]
[image]
Description: I manage to get all the available button work, but I can't make the central strip work, even if it seems it is detected:
What I have available:

```
xsetwacom --list devices
Tablet Monitor stylus           	id: 10	type: STYLUS    
Tablet Monitor Pad pad          	id: 11	type: PAD       
Tablet Monitor Touch Strip pad  	id: 12	type: PAD       
Tablet Monitor Dial pad         	id: 13	type: PAD   
```    
I have scripts to use when i connect the tablet (I'm on a laptop PC, I don't always lug the tablet with me)

```
#!/bin/bash
# this one maps the pen on its own monitor and not all two 
echo "penna su tablet"
xinput map-to-output "Tablet Monitor stylus" DP-1-3
```
This one maps all buttons I was able to get working (there are other five buttons below 13 but I guess they are reserved for OSD setup thus not available)
```
#!/bin/sh
xsetwacom --set 'Tablet Monitor Pad pad' Button 1 "key plus"
xsetwacom --set 'Tablet Monitor Pad pad' Button 2 "key +ISO_Level3_Shift +8"
xsetwacom --set 'Tablet Monitor Pad pad' Button 3 "key +ctrl +z -z -ctrl"
xsetwacom --set 'Tablet Monitor Pad pad' Button 8 "key +ISO_Level3_Shift +9"
xsetwacom --set 'Tablet Monitor Pad pad' Button 9 "key minus"
xsetwacom --set 'Tablet Monitor Pad pad' Button 10 "key e"
xsetwacom --set 'Tablet Monitor Pad pad' Button 11 "key m"
xsetwacom --set 'Tablet Monitor Pad pad' Button 12 "key ctrl"
xsetwacom --set 'Tablet Monitor Pad pad' Button 13 "key shift"
```
I'm not aware if there's any setup similar to buttons and I tried these ones, to no avail:
(found on other issues after searching for keyword "huion kamvas 16 touch stripe how to|not work")
```
xsetwacom --set "Tablet Monitor Touch Strip pad" StripLeftUp "key plus"
xsetwacom --set "Tablet Monitor Touch Strip pad" StripLeftDown "key minus"
```
Available to other kind of test etc. I'm a basic user so I might need some newbie-proof explanation, I apologize.

superuser-miguel, 2023-12-28:
I know this is a random question, Do you have drift while using the Pen? The further away you get from the center of the screen if your pointer further away from the Pen?

matitaelettrica, 2023-12-29:
> I know this is a random question, Do you have drift while using the Pen? [...]

I don't have this problem. Apart from the issue I describe (touch strip not working) the tablet works without problems.

JoseExposito, 2024-05-03:
Hi @matitaelettrica ,

Please try with the latest release ([v13](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v13)) as it should send the right events for touch strips.

You'll need to wait for the next libinput release for the mapping to be accurate (i.e., expect some weird values at the moment), but the overall experience should be a bit better.

JoseExposito, 2024-05-03:
Oh! This is the relevant merge request in libinput: https://gitlab.freedesktop.org/libinput/libinput/-/merge_requests/989

The change will be available in libinput v1.26

matitaelettrica, 2024-06-14:
Hi @JoseExposito,
(apologies for late reply)
I uninstalled old version and installed new version of drivers, v13.
I also  had some update in the system, now I'm on linux Mint 21.3 Cinnamon and kernel 5.15.0-112-generic

Sadly I don't have any kind of response from the touch strip, even a wrong one, but I'm suspecting that's because I don't know the right kind of code to assign the strip behaviour in my script. I'm using these lines of code:

```
xsetwacom --set "Tablet Monitor Touch Strip pad" "StripLeftUp" "1" "key +plus -plus"
xsetwacom --set "Tablet Monitor Touch Strip pad" "StripLeftDown" "2" "key +minus -minus"
```
which I found in many searches about this specific topic. [Buttons work as desired.]

I'll wait for the next release.

JoseExposito, 2024-06-14:
Hi @matitaelettrica 

If you run `sudo libinput record` and select your tablet's touchstrip event number, do you see any output in the terminal when you touch the touch strips? If you can, attach the output and I'll have a look.

If your device sends Rx and Ry events, then you should be able to configure the touch strips (and buttons) from GNOME or KDE settings.

You might need to make minimal changes to libwacom's descriptor file:
https://github.com/linuxwacom/libwacom/blob/master/data/huion-kamvas-16-gs1562.tablet

But I can help you with those :)

matitaelettrica, 2024-06-16:
Hi @JoseExposito 

This is the output of a single movement, from top to bottom of strip, with `sudo libinput`:
[libinput-touchstrip.txt](https://github.com/user-attachments/files/15859253/libinput-touchstrip.txt)

To my untrained eye seems that the event is at least recognized? 

Also to my untrained eye the libwacom descriptor for my tablet seems more this one: 
https://github.com/linuxwacom/libwacom/blob/master/data/huion-kamvas-162019.tablet
the `numstrips` parameter is set at 1 which is what I have on the tablet (one single vertical strip)

I'm on Cinnamon, which has origin on Gnome2 if I recall correctly, I'll make some research on this side too.

Thanks again for the pointers!

matitaelettrica, 2024-07-04:
Thanks again! And sorry again for being late.

The `libwacom` on this system is quite ancient... so to me is waiting for the new version of Linux Mint -- or trying to upgrade the library, a thing I'm not sure I have the skillset to correctly approach. 

(Also, I found on my system all the *.tablet descriptor files are under `/usb/share/libwacom/ ` instead of `/etc/libwacom/` , might this give problems?)

JoseExposito, 2024-07-05:
> Also, I found on my system all the *.tablet descriptor files are under /usr/share/libwacom/  instead of /etc/libwacom/ , might this give problems?

That's OK, `/usr/share/libwacom` contains the files installed by default. `/etc/libwacom` is mainly used to test new files.