DIGImend/digimend-kernel-drivers #673: dkms_install throws away errors: `Error! Your kernel headers for kernel 6.5.12_1 cannot be found at /lib/modules/6.5.12_1/build or /lib/modules/6.5.12_1/source.`

LinArcX, 2023-12-01:
I'm using VoidLinux and I downloaded v11.zip from release section.

when I run: `sudo make dkms_install`, I will get these errors:

```
dkms add .
Creating symlink /var/lib/dkms/digimend/11/source -> /usr/src/digimend-11
dkms build digimend/11
Sign command: /lib/modules/6.5.12_1/build/scripts/sign-file
Binary /lib/modules/6.5.12_1/build/scripts/sign-file not found, modules won't be signed
Error! Your kernel headers for kernel 6.5.12_1 cannot be found at /lib/modules/6.5.12_1/build or /lib/modules/6.5.12_1/source.
Please install the linux-headers-6.5.12_1 package or use the --kernelsourcedir option to tell DKMS where it's located.
make: *** [Makefile:101: dkms_modules_install] Error 1
```

Some informatin:
```
↳ uname -a
Linux gnu_linux 6.5.12_1 #1 SMP PREEMPT_DYNAMIC Mon Nov 20 18:31:50 UTC 2023 x86_64 GNU/Linux
```
Also `linux6.5-headers-6.5.13_1` package is installed on my machine.
Any idea?

LinArcX, 2023-12-01:
Ok, i solved that issue. it was because my system was not update. after updating the kernel and run again the same command, i will receive new error:
```
dkms add .
Creating symlink /var/lib/dkms/digimend/11/source -> /usr/src/digimend-11
dkms build digimend/11
Sign command: /lib/modules/6.5.13_1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j8 KERNELRELEASE=6.5.13_1 KVERSION=6.5.13_1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.5.13_1 (x86_64)
Consult /var/lib/dkms/digimend/11/build/make.log for more information.
make: *** [Makefile:101: dkms_modules_install] Error 10
```

And this is the content of `/var/lib/dkms/digimend/11/build/make.log`:
```
DKMS make.log for digimend-11 for kernel 6.5.13_1 (x86_64)
Fri Dec  1 07:54:18 PM CET 2023
make[1]: Entering directory '/var/lib/dkms/digimend/11/build'
make -C /lib/modules/6.5.13_1/build M=/var/lib/dkms/digimend/11/build modules
make[2]: Entering directory '/usr/src/kernel-headers-6.5.13_1'
  CC [M]  /var/lib/dkms/digimend/11/build/hid-kye.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-uclogic-core.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-uclogic-rdesc.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-uclogic-params.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-polostar.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-viewsonic.o
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c: In function 'uclogic_params_init':
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c:1227:17: error: implicit declaration of function 'hid_is_using_ll_driver' [-Werror=implicit-function-declaration]
 1227 |             || !hid_is_using_ll_driver(hdev, &usb_hid_driver)
      |                 ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c:1227:47: error: 'usb_hid_driver' undeclared (first use in this function); did you mean 'to_hid_driver'?
 1227 |             || !hid_is_using_ll_driver(hdev, &usb_hid_driver)
      |                                               ^~~~~~~~~~~~~~
      |                                               to_hid_driver
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c:1227:47: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c:1349:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1349 |                 if (bNumInterfaces != 3) {
      |                    ^
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c:1367:9: note: here
 1367 |         case VID_PID(USB_VENDOR_ID_HUION,
      |         ^~~~
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:243: /var/lib/dkms/digimend/11/build/hid-uclogic-params.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/kernel-headers-6.5.13_1/Makefile:2034: /var/lib/dkms/digimend/11/build] Error 2
make[2]: *** [Makefile:234: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/kernel-headers-6.5.13_1'
make[1]: *** [Makefile:25: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/digimend/11/build'
```

matous-volf, 2023-12-25:
I am experiencing exactly the same problem on Pop!_OS with kernel version `6.6.6-76060606-generic`. Did you manage to solve it or find additional info?

Is it possible that the issue is similar to [this one](https://bugs.launchpad.net/ubuntu/+source/digimend-dkms/+bug/2011546) – the absence or a wrong setting of some kernel parameters (like `CONFIG_USB`)?

LinArcX, 2023-12-25:
@matous-volf 

I switched to: https://opentabletdriver.net/

and it's amazing for me.

matous-volf, 2023-12-29:
That works for me too. Thanks!