DIGImend/digimend-kernel-drivers #677: release v11 fails to install or compile (implicit function declaration error)

seine-axelenz, 2024-01-04:
i am running Pop!_OS 22.04 (ubuntu derivative), kernel version 6.6.6-76060606-generic
installing release v11 fails from both the ubuntu repositories and the provided .deb, and also when compiling from source.

Kernel headers and dkms are installed, running `sudo make dkms_install` results in the following error

```
dkms add .
Creating symlink /var/lib/dkms/digimend/11/source -> /usr/src/digimend-11
dkms build digimend/11

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=6.6.6-76060606-generic KVERSION=6.6.6-76060606-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for digimend: 11 not found
Error! Bad return status for module build on kernel: 6.6.6-76060606-generic (x86_64)
Consult /var/lib/dkms/digimend/11/build/make.log for more information.
make: *** [Makefile:101: dkms_modules_install] Error 10

```

and the log file at /var/lib/dkms/digimend/11/build/make.log: 

```
DKMS make.log for digimend-11 for kernel 6.6.6-76060606-generic (x86_64)
Thu Jan  4 03:49:57 PM CET 2024
make[1]: Entering directory '/var/lib/dkms/digimend/11/build'
make -C /lib/modules/6.6.6-76060606-generic/build M=/var/lib/dkms/digimend/11/build modules
make[2]: Entering directory '/usr/src/linux-headers-6.6.6-76060606-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  CC [M]  /var/lib/dkms/digimend/11/build/hid-kye.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-uclogic-core.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-uclogic-rdesc.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-uclogic-params.o
  CC [M]  /var/lib/dkms/digimend/11/build/hid-polostar.o
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c: In function ‘uclogic_params_init’:
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c:1227:17: error: implicit declaration of function ‘hid_is_using_ll_driver’ [-Werror=implicit-function-declaration]
 1227 |             || !hid_is_using_ll_driver(hdev, &usb_hid_driver)
      |                 ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c:1227:47: error: ‘usb_hid_driver’ undeclared (first use in this function); did you mean ‘to_hid_driver’?
 1227 |             || !hid_is_using_ll_driver(hdev, &usb_hid_driver)
      |                                               ^~~~~~~~~~~~~~
      |                                               to_hid_driver
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c:1227:47: note: each undeclared identifier is reported only once for each function it appears in
  CC [M]  /var/lib/dkms/digimend/11/build/hid-viewsonic.o
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c:1349:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1349 |                 if (bNumInterfaces != 3) {
      |                    ^
/var/lib/dkms/digimend/11/build/hid-uclogic-params.c:1367:9: note: here
 1367 |         case VID_PID(USB_VENDOR_ID_HUION,
      |         ^~~~
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:243: /var/lib/dkms/digimend/11/build/hid-uclogic-params.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-6.6.6-76060606-generic/Makefile:1919: /var/lib/dkms/digimend/11/build] Error 2
make[2]: *** [Makefile:234: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.6.6-76060606-generic'
make[1]: *** [Makefile:25: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/digimend/11/build'
```

it would seem like the compiler doesn't like a certain implicit function declaration, however i am not well-versed enough in c/++ or linux' inner workings to get further than this

ktrace, 2024-01-05:
Look like "not build" with kernel >6.3
Next commit after release is a cure:
https://github.com/DIGImend/digimend-kernel-drivers/commit/eca6e1b701bffb80a293234a485ebf6b4bc85562.patch

JoseExposito, 2024-02-24:
There is a new release available, hopefully it'll solve your issue.

Closing the bug report because, as @ktrace mentioned, the fix is already merged.

closed 2024-02-24 by JoseExposito