DIGImend/digimend-kernel-drivers #697: hid-uclogic should set Pen or Digitizer, depending on usage

whot, 2024-06-19:
Filing this mostly so there's a record somehere.

The report descriptor fixup `uclogic_rdesc_v1_pen_template_arr` sets `Usage (Digitizer)` but the kernel differs between "Digitizer" (for external tablets, `INPUT_PROP_POINTER`) and "Pen" for display tablets (`INPUT_PROP_DIRECT`). Userspace somewhat relies on `INPUT_PROP_DIRECT` to know if we're dealing with a direct-input device (see [libinput!1019](https://gitlab.freedesktop.org/libinput/libinput/-/merge_requests/1019) for a workaround).

Ideally uclogic would set pen vs depending on the device, though I understand that's not exactly trivial in the current implementation.

JoseExposito, 2024-06-19:
As far as I know, there is no information in the tablet or pen descriptor (`uclogic_params_pen_init_v2`) to differentiate them. Here are a couple of examples of `INPUT_PROP_DIRECT` devices:

- https://github.com/DIGImend/digimend-kernel-drivers/issues/633
  S C8 13 03 80 E5 00 18 81 00 FF 1F D8 13 03 05 00 01 08 00 40
- https://github.com/DIGImend/digimend-kernel-drivers/issues/560
  S C8 13 03 BA 53 01 88 BA 00 FF 1F D8 13 03 00 00 00 08 00 40

The only way to differentiate them is creating a list of devices with screen in the kernel... But that's already done in libwacom, so your solution works pretty well.
If a kernel fix is really needed, I can work on one, but I don't think it is worth it.

Leaving this bug open, as it is not fixed here, but most likely, we won't see a fix in the kernel any time soon.

whot, 2024-06-19:
we already map the firmware to the uniq here, right? so it would be relatively trivial to generate a firmware id to internal/external mapping from libwacom and add that here. But yeah, ETIME and EMOTIVATION :)