DIGImend/digimend-kernel-drivers #704: Mutter complains about multiple tablets

khoidauminh, 2024-10-30:
My tablet is a Gaomon 1060pro and it works with the stylus and its buttons

`sudo libinput list-devices` still lists the problematic device entry:
```
Device:           GAOMON Gaomon Tablet_1060Pro
Kernel:           /dev/input/event21
Group:            5
Seat:             seat0, default
Size:             164x164mm
Capabilities:     tablet 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   none
Rotation:         n/a

```

The entry that works:
```
Device:           GAOMON Gaomon Tablet_1060Pro Pen
Kernel:           /dev/input/event7
Group:            5
Seat:             seat0, default
Size:             254x159mm
Capabilities:     tablet 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   none
Rotation:         n/a
```

This causes mutter to complain about multiple tablets:
```
Thg 10 30 18:00:47 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:00:57 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:00:57 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:00:57 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:00:57 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:27 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:27 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:28 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:28 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:28 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:28 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:33 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:33 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:33 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:33 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:36 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:36 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:36 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
Thg 10 30 18:01:36 kpc gnome-shell[1889]: (../mutter-dynamic-buffering/src/wayland/meta-wayland-tablet-seat.c:520):meta_wayland_tablet_seat_lookup_paired_tablet: runtime check failed: (!devices->next)
[... 271 lines cut ...]
Disable-w-trackpointing: n/a
Accel profiles:   n/a
Rotation:         0.0

Device:           HP WMI hotkeys
Kernel:           /dev/input/event9
Group:            9
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   n/a
Rotation:         0.0
```

</details>

debug686, 2025-03-13:
Is this why I see my tablet listed 3 times in the Gnome Wacom Tablet settings?

khoidauminh, 2025-03-14:
@debug686 Libinput creates 3 devices for my tablet so it shows up in the Gnome settings 3 times. Probably the case for you too
```bash
kdm@kpc ~> sudo libinput list-devices | grep GAOMON
event19 - GAOMON Gaomon Tablet_1060Pro Touch Strip: libinput bug: missing tablet capabilities: pen btn-stylus resolution. Ignoring this device.
event20 - GAOMON Gaomon Tablet_1060Pro Dial: libinput bug: missing tablet capabilities: pen btn-stylus resolution. Ignoring this device.
Device:           GAOMON Gaomon Tablet_1060Pro Pen
Device:           GAOMON Gaomon Tablet_1060Pro Pad
Device:           GAOMON Gaomon Tablet_1060Pro Group Buttons
```

With the hack in-place it still happens btw