DIGImend/digimend-kernel-drivers #706: Build error: linux 6.12 renames asm/unaligned.h

MrElendig, 2024-11-23:
as of kernel 6.12-rc2 asm/unaligned.h  have been moved to linux/unaligned.h

https://lkml.org/lkml/2024/10/6/423

[buildlog.txt](https://github.com/user-attachments/files/17882737/buildlog.txt)

Stroopwafe1, 2024-11-25:
I also had this problem and made a PR #707 . Let's hope it's ok

JoseExposito, 2025-01-02:
Fixed by https://github.com/DIGImend/digimend-kernel-drivers/pull/707

closed 2025-01-02 by JoseExposito

vinassefranche, 2025-05-05:
Hi here!

I still have the issue when trying to install the kernel 6.12.25 with mainline on my Ubuntu 24.
Is there anything I need to do before doing the kernel install to make it work?

Thanks in advance for your answers

MrElendig, 2025-05-05:
> Hi here!
> 
> I still have the issue when trying to install the kernel 6.12.25 with mainline on my Ubuntu 24.
> Is there anything I need to do before doing the kernel install to make it work?
> 
> Thanks in advance for your answers

Building from HEAD should work

vinassefranche, 2025-05-05:
> > Hi here!
> > I still have the issue when trying to install the kernel 6.12.25 with mainline on my Ubuntu 24.
> > Is there anything I need to do before doing the kernel install to make it work?
> > Thanks in advance for your answers
> 
> Building from HEAD should work

You mean that I should follow the [Installing from source manually](https://github.com/DIGImend/digimend-kernel-drivers?tab=readme-ov-file#installing-from-source-manually) steps from the HEAD of the master branch?

I can try this but it's weird that I'd have to do that manually no? The normal install should work smoothly, I think.

MrElendig, 2025-05-05:
> You mean that I should follow the [Installing from source manually](https://github.com/DIGImend/digimend-kernel-drivers?tab=readme-ov-file#installing-from-source-manually) steps from the HEAD of the [...]

The latest release(v13) is a year old and doesn't include the fix.

Alternative to building from HEAD is to cherrypick the commits which fixes the issue.

vinassefranche, 2025-05-05:
> The latest release(v13) is a year old and doesn't include the fix.

Ah ok. Do you know if there's a release planned soon? If it's a matter of weeks or a few months, I'm fine with waiting for the official install instead of trying to build it myself.

Anyway, thanks for your answers!

MrElendig, 2025-05-05:
> > The latest release(v13) is a year old and doesn't include the fix.
> 
> Ah ok. Do you know if there's a release planned soon? If it's a matter of weeks or a few months, I'm fine with waiting for the official install instead of trying to build it myself.
> 
> Anyway, thanks for your answers!

How are you currently installing the driver?

vinassefranche, 2025-05-05:
> How are you currently installing the driver?

Not completely sure to be honest but I think it was either installed by Ubuntu when I installed the OS or with a kernel I installed with mainline.
I might have installed it with `apt` someday if it was among the installation steps of another package but I don't remember it.

vinassefranche, 2025-09-01:
@MrElendig Sorry to bother you again but I still have no fix to install new kernels on my machine because of this issue.

I'm currently running my machine on the kernel 6.11.0-29 and I have the drivers running properly. I'm trying to install the kernel 6.16.4 (or any kernel starting from 6.12) and I have the same issue as described in the description of this issue.
I understood I need to build the drivers from source manually but when I run `sudo make dkms_install` or `sudo make dkms_install`, it's building the drivers for my current kernel version (6.11.0-29) and not the ones for the more recent kernel version.

Am I doing something wrong? I once tried to boot with new kernel without the drivers and the boot completely failed and I feared I had to re-install everything so I hope it's not the solution to my issue :-)

Thanks in advance for all the support you can provide me on this 🙏🏼

vinassefranche, 2025-09-02:
> > [@MrElendig](https://github.com/MrElendig) Sorry to bother you again but I still have no fix to install new kernels on my machine because of this issue.
> 
> You need the kernel headers for 6.14.4 and you might have to explicitly tell dkms which kernel to build for. See https://wiki.archlinux.org/title/Dynamic_Kernel_Module_Support#Rebuild_modules and dkms(8)

I'll give it a try. Thanks for the answer!