DIGImend/digimend-kernel-drivers #716: Issue with Linux Kernal 6.15

BIGBEASTISHANK, 2025-07-12:
My Huion H1060P stoped working when i checked the update logs after kernal update, digiend-kernel-drivers gave error and stoped working. Here is the error

```
╭─ishank@ArchBeast /home/ishank/Temp/digimend-kernel-drivers  <master>
╰─$ makepkg -si
==> ERROR: PKGBUILD does not exist.
╭─ishank@ArchBeast /home/ishank/Temp/digimend-kernel-drivers  <master>
╰─$ cmake                                                                                                                                                                               ↵ 12
Usage

  cmake [options] <path-to-source>
  cmake [options] <path-to-existing-build>
  cmake [options] -S <path-to-source> -B <path-to-build>

Specify a source directory to (re-)generate a build system for it in the
current working directory.  Specify an existing build directory to
re-generate its build system.

Run 'cmake --help' for more information.

╭─ishank@ArchBeast /home/ishank/Temp/digimend-kernel-drivers  <master>
╰─$ make -f Makefile
make -C /lib/modules/6.15.6-arch1-1/build M=/home/ishank/Temp/digimend-kernel-drivers modules
make[1]: Entering directory '/usr/lib/modules/6.15.6-arch1-1/build'
make[2]: Entering directory '/home/ishank/Temp/digimend-kernel-drivers'
  CC [M]  hid-kye.o
  CC [M]  hid-uclogic-core.o
hid-uclogic-core.c: In function ‘uclogic_remove’:
hid-uclogic-core.c:494:9: error: implicit declaration of function ‘del_timer_sync’ [-Wimplicit-function-declaration]
  494 |         del_timer_sync(&drvdata->inrange_timer);
      |         ^~~~~~~~~~~~~~
make[4]: *** [/usr/lib/modules/6.15.6-arch1-1/build/scripts/Makefile.build:203: hid-uclogic-core.o] Error 1
make[3]: *** [/usr/lib/modules/6.15.6-arch1-1/build/Makefile:2003: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.6-arch1-1/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/home/ishank/Temp/digimend-kernel-drivers'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.6-arch1-1/build'
make: *** [Makefile:26: modules] Error 2
```

```
╭─ishank@ArchBeast /home/ishank
╰─$ paru -S digimend-kernel-drivers-dkms-git
:: Resolving dependencies...
:: Calculating conflicts...
:: Calculating inner conflicts...

Aur (1) digimend-kernel-drivers-dkms-git-10.r0.gec5c16d-2

:: Proceed to review? [Y/n]:

:: Downloading PKGBUILDs...
 PKGBUILDs up to date
 nothing new to review
fetching devel info...
==> Making package: digimend-kernel-drivers-dkms-git 10.r0.gec5c16d-2 (Sunday 13 July 2025 01:07:26 AM)
==> Retrieving sources...
  -> Updating digimend-kernel-drivers git repo...
==> Validating source files with md5sums...
    digimend-kernel-drivers ... Skipped
==> Making package: digimend-kernel-drivers-dkms-git 10.r0.gec5c16d-2 (Sunday 13 July 2025 01:07:28 AM)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating digimend-kernel-drivers git repo...
==> Validating source files with md5sums...
    digimend-kernel-drivers ... Skipped
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of digimend-kernel-drivers git repo...
[... 105 lines cut ...]
  CC [M]  hid-viewsonic.o
hid-uclogic-params.c: In function ‘uclogic_params_init’:
hid-uclogic-params.c:1707:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1707 |                 if (bNumInterfaces != 3) {
      |                    ^
hid-uclogic-params.c:1725:9: note: here
 1725 |         case VID_PID(USB_VENDOR_ID_HUION,
      |         ^~~~
hid-uclogic-core.c: In function ‘uclogic_remove’:
hid-uclogic-core.c:494:9: error: implicit declaration of function ‘del_timer_sync’ [-Wimplicit-function-declaration]
  494 |         del_timer_sync(&drvdata->inrange_timer);
      |         ^~~~~~~~~~~~~~
make[4]: *** [/usr/lib/modules/6.15.6-arch1-1/build/scripts/Makefile.build:203: hid-uclogic-core.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/lib/modules/6.15.6-arch1-1/build/Makefile:2003: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.6-arch1-1/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/digimend-kernel-drivers/11.r42.gf3c7c7f/build'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.6-arch1-1/build'
make: *** [Makefile:26: modules] Error 2

# exit code: 2
# elapsed time: 00:00:03
----------------------------------------------------------------
```

talpadk, 2025-07-27:
There is already a merge request for a fix https://github.com/DIGImend/digimend-kernel-drivers/pull/714

I build the fork https://github.com/tokenicrat/digimend-kernel-drivers successfully on 6.15.8

BIGBEASTISHANK, 2025-07-29:
> There is already a merge request for a fix [#714](https://github.com/DIGImend/digimend-kernel-drivers/pull/714)
> 
> I build the fork https://github.com/tokenicrat/digimend-kernel-drivers successfully on 6.15.8

Yaa bud I remember, The pr you created had some issues I created a pr on your pr please check and i am closing this issue

closed 2025-07-29 by BIGBEASTISHANK